HorizonRobotics/RoboOrchardLab #2: Add evaluation callback and best checkpoint saving during training

Fanqyu, 2025-07-10:
Hi Horizon Robotics Team, 

Thank you for your excellent open-source baseline for RoboTwin!

I’ve been going through the training pipeline and noticed that the current implementation does not include any evaluation step (e.g., validation callback) during training, which means there’s no mechanism to save the best-performing checkpoints based on metrics. I'm wondering if you have any plans to support this functionality, or if contributions in this direction would be welcome.

In addition, I'm reproducing your results using the default configuration on a single 8GB RTX 4060 GPU. The training runs for 100,000 steps and saves checkpoints every 4000 steps. According to the current logic, only the last three checkpoints are retained. I'm curious whether this checkpoint retention strategy was chosen based on evaluation considerations, or if it’s simply a placeholder/default setting.

Looking forward to your response. I'd be happy to help contribute if this is something the team would like to support!

Best regards,
Qingyu

Fanqyu, 2025-07-11:
Thanks for your kind reply. This info is very helpful.

Since you mentioned that there is no significant performance difference between the last checkpoint and the best checkpoint, I’d like to know how you determine the training step length. Are there any tricks or best practices you could share regarding this? Looking forward to your insights!

Also, if I were to go through the training pipeline using domain-randomized training data, are there any changes needed, such as adjusting max step lengths or other hyperparameters?

linxuewu, 2025-07-11:
Our empirical findings indicate that:​​
1. ​​For single-task training​​, 100 demonstrations per task with 100k steps are sufficient for convergence.
​​2. For multi-task training​​ (e.g., 18 tasks), 100 demonstrations per task with 200k steps achieve convergence.
​​3. For simple, short-horizon tasks​​ (e.g., 'empty cup place'), the step count can be appropriately reduced - for instance, from 100k to 60k steps.

​​When task complexity increases​​ or when more training data becomes available, the step count should be correspondingly increased.

These hyperparamters(max_step/lr/lr_scheduler) represent optimized empirical settings. If your application scenario differs significantly, manual tuning will be necessary. You may also need to test with intermediate checkpoints to achieve better performance sometimes.

Fanqyu, 2025-07-14:
Hi, thank you so much for your detailed and thoughtful responses — they’ve been incredibly helpful!

I have a couple of follow-up questions:

1. Is `scale_shift` used as a form of normalization in the pipeline? And is it determined solely based on the training data?

2. I trained the `place_cup_empty` task using 100 domain-randomized demonstrations, following your suggested setup. I achieved a best result of 35%, which aligns with the performance reported in the SEM arXiv paper. However, this is still quite far from the near-100% accuracy mentioned in the CVPR challenge results. Are there additional factors I should consider adjusting — such as increasing the number of demonstrations or tuning other hyperparameters?

Thanks again for your continued support!

Fanqyu, 2025-07-15:
Thank you for the clarification and the detailed information — much appreciated!

closed 2025-07-15 by Fanqyu