HorizonRobotics/RoboOrchardLab #3: A question about GlobalStep and the Checkpoint Resumption

Chloe-Yan-3465, 2025-07-12:
[issue.md](https://github.com/user-attachments/files/21194952/issue.md)

Chloe-Yan-3465, 2025-07-13:
We evaluated the `place_empty_cup` task and obtained a 52.0% success rate (terminal screenshot attached).

We'd like to ask whether the camera `depth` configuration wasn't specifically set up in the `eval` process. The `eval` only takes the `camera_config.yml` as input, which doesn't specify `rgbd` settings - it only includes height and width parameters. During evaluation, we noticed many data points lack `depth` information (as shown in the terminal output).

[image]

 even though we set `depth=True` during data collection. We suspect that the evaluation might not be fully utilizing depth information, which could explain why we can't reproduce the 85% success rate reported in the paper. Where can we get the depth? (Below is the ``sem_policy/deploy_policy.py``,we make some adjustment of ``camera_data`` regarding code)Could you please provide some guidance on this?

```
def encode_obs(observation, data_transforms, instruction):
    images = []
    depths = []
    t_world2cam = []
    intrinsics = []
    joint_state = []

    missing_depth_cameras = []

    for cam_name, camera_data in observation["observation"].items():
        # 添加 RGB 图像
        images.append(camera_data["rgb"])

        # 处理 depth，有就用，没有就构造全零占位
        if "depth" in camera_data:
            depths.append(camera_data["depth"] / 1000)  # mm -> m
        else:
            print(f"[Warning] Camera '{cam_name}' has no depth, inserting zero placeholder.")
            h, w = camera_data["rgb"].shape[:2]
            fake_depth = np.zeros((h, w), dtype=np.float32)
            depths.append(fake_depth)
            missing_depth_cameras.append(cam_name)

        # 添加相机外参
        _ext = np.eye(4, dtype=np.float32)
        _ext[:3] = camera_data["extrinsic_cv"]
        t_world2cam.append(_ext)

        # 添加相机内参
        _int = np.eye(4, dtype=np.float32)
        _int[:3, :3] = camera_data["intrinsic_cv"]
        intrinsics.append(_int)
```

Thanks in advance for your help! we really admire this work and want to reproduce it faithfully!

wzmsltw, 2025-07-14:
Hi,  you can modify the task config to enable depth simulation. For example, if you eval model with task_config/demo_clean.yml, you should modify data_type/depth from false to true.

Chloe-Yan-3465, 2025-07-14:
Okay! Thanks a lot for it !

This one ,right ?
```
data_type:
  rgb: true
  third_view: false
  depth: true #7.9 sem
```

Sorry I've made a silly mistake (apologies for the noise >< ). I forgot to change `task_config=demo_clean` in `sem_policy/eval.sh` , what I actually used to collect data is `sem_place_empty_cup.yml` . 

Really sorry for the bothering, and I sincerely appreciate your patience and support!

I'll go ahead and close this issue.

[image]