HorizonRobotics/RoboOrchardLab #7: other tasks

Winstongh, 2025-07-22:
Hello , I want to ask how to calculate scale_shift in other tasks

linxuewu, 2025-07-22:
You need to iterate through the training set, collect all the `joint_state`, and then do statistics.

```
max_v = joint_state.reshape(-1, 7).max(axis=0)
min_v = joint_state.reshape(-1, 7).min(axis=0)
scale = (max_v - min_v) / 2
shift = (max_v + min_v) / 2
```

Muyun99, 2025-07-22:
Some code for reference, add this function in dataset can calculate the `scale` and `shift`
```python

class Dataset(BaseLmdbManipulationDataset):
    ...
    def get_scale_shift(self):
        joint_state_list = []
        for lmdb_index in range(len(self.idx_lmdbs)):
            for episode_index in range(len(self.idx_lmdbs[lmdb_index]) - 1):
                idx_data = self.idx_lmdbs[lmdb_index][episode_index]
                uuid = idx_data["uuid"]

                joint_state = self.meta_lmdbs[lmdb_index][
                    f"{uuid}/observation/robot_state/joint_positions"
                ]
                joint_state_list.append(joint_state)
                num_dof = joint_state.shape[-1]

        joint_state_list = np.concatenate(joint_state_list, axis=0).reshape(
            -1, num_dof
        )
        max_v = joint_state_list.max(axis=0)
        min_v = joint_state_list.min(axis=0)
        scale = (max_v - min_v) / 2
        shift = (max_v + min_v) / 2

        print("-" * 18, "The Analyse is here", "-" * 18)
        print(f"scale is {scale}")
        print(f"shift is {shift}")
        for i in range(num_dof):
            print(f"[{scale[i]}, {shift[i]}],")

dataset = Dataset()
dataset.get_scale_shift()
```

bbbbbbbbly, 2025-10-16:
Hello, I added a task for both arms to reach the block, but for some reason the left arm isn't reaching the center point of the block very often. My data shows it should be reaching the center point perfectly. Do I also need to calculate this scale and shift? And once I calculate it, how do I apply it?

https://github.com/user-attachments/assets/4c6f03be-4e61-406e-823d-6ca7f6455a7f