HorizonRobotics/RoboOrchardLab #8: sem data processing

yaocwbcgs, 2025-07-24:
Could you please help me with an issue? For the SEM data processing part, it seems that the CVPR-Challenge-2025-Round2 branch isn't working properly. Would you mind giving me some guidance on this? Thank you so much!

bbbbbbbbly, 2025-10-20:
hello, I can run this command successfully, but I don't know how to construct multi task data.
```
 /home/k/da/t/RoboOrchardLab  master !3 ?11  python robo_orchard_lab/dataset/robotwin/robotwin_packer.py --input_path /home/kemove/data/RoboTwin/data --output_path "projects/sem/robotwin/data/lmdb_reach_50" --task_names reach_different_blocks --config_name demo_clean_arx_eef`
```
could you please me some advice how to do the multi task trainging in data process.

AlanHuangT, 2026-04-07:
This issue was closed because it has been inactive