HorizonRobotics/RoboOrchardLab #10: ckpt_rename.py is missing

luo2413669045, 2025-07-29:
You are in the readme mentioned that you use python tools/ckpt_rename.py ckpt/groundingdino_swint_ogc_mmdet-822d7e9d.pth --output ./ckpt command, but the ckpt_rename.py file seems to be missing in the project.

linxuewu, 2025-07-29:
​Sorry for missing this file.

You can use the following script instead: https://github.com/HorizonRobotics/robo_orchard_lab/blob/master/projects/bip3d_grounding/tools/ckpt_rename.py.

luo2413669045, 2025-07-29:
Thank you very much for taking the time to answer my question.

I had also noticed this file earlier and tried using it to rename the files. However, it doesn't seem to correspond to the renaming process for SAM, as it doesn't map the network layers in the checkpoint to the correct names. Even after renaming with it, I still encounter the issue described below.

Once again, I truly appreciate your response.

----------------------------------------------------------------------------------
Rank[0/1] 07/29/2025 15:03:08 WARNING robotwin_lmdb_dataset.py:127 | dataset T_base2world is not set, use default.
Rank[0/1] 07/29/2025 15:03:08 WARNING robotwin_lmdb_dataset.py:127 | dataset T_base2world is not set, use default.
Rank[0/1] 07/29/2025 15:03:08 INFO base_lmdb_dataset.py:186 | dataset length: 7571, number of episode: 10
Rank[0/1] 07/29/2025 15:03:08 INFO utils.py:61 | load checkpoint: ./ckpt/groundingdino_swint_ogc_mmdet-822d7e9d-rename.pth
Rank[0/1] 07/29/2025 15:03:08 INFO utils.py:75 | num of missing_keys: 536,num of unexpected_keys: 627
Rank[0/1] 07/29/2025 15:03:08 INFO utils.py:79 | missing_keys:
 ['decoder.robot_encoder.input_fc.0.weight', 'decoder.robot_encoder.input_fc.0.bias', 'decoder.robot_encoder.input_fc.2.weight', 'decoder.robot_encoder.input_fc.2.bias', 'decoder.robot_encoder.input_fc.4.weight', 'decoder.robot_encoder.input_fc.4.bias', 'decoder.robot_encoder.input_fc.5.weight', 'decoder.robot_encoder.input_fc.5.bias', 'decoder.robot_encoder.input_fc.7.weight', 'decoder.robot_encoder.input_fc.7.bias', 'decoder.robot_encoder.input_fc.9.weight', 'decoder.robot_encoder.input_fc.9.bias', 'decoder.robot_encoder.input_fc.10.weight', 'decoder.robot_encoder.input_fc.10.bias', 'decoder.robot_encoder.layers.0.weight', 'decoder.robot_encoder.layers.1.q_proj.weight', 'decoder.robot_encoder.layers.1.q_proj.bias', 'decoder.robot_encoder.layers.1.k_proj.weight', 'decoder.robot_encoder.layers.1.v_proj.weight', 'decoder.robot_encoder.layers.1.v_proj.bias', 'decoder.robot_encoder.layers.1.proj.weight', 'decoder.robot_encoder.layers.1.proj.bias', 'decoder.robot_encoder.layers.1.position_encoder.freqs', 'decoder.robot_encoder.layers.1.position_encoder.mlp.0.weight', 'decoder.robot_encoder.layers.1.position_encoder.mlp.0.bias', 'decoder.robot_encoder.layers.1.position_encoder.mlp.2.weight', 'decoder.robot_encoder.layers.1.position_encoder.mlp.2.bias', 'decoder.robot_encoder.layers.4.weight', 'decoder.robot_encoder.layers.5.layers.0.0.weight', 'decoder.robot_encoder.layers.5.layers.0.0.bias', 'decoder.robot_encoder.layers.5.layers.1.weight', 'decoder.robot_encoder.layers.5.layers.1.bias', 'decoder.robot_encoder.layers.6.weight', 'decoder.robot_encoder.layers.7.q_proj.weight', 'decoder.robot_encoder.layers.7.q_proj.bias', 'decoder.robot_encoder.layers.7.k_proj.weight', 'decoder.robot_encoder.layers.7.v_proj.weight', 'decoder.robot_encoder.layers.7.v_proj.bias', 'decoder.robot_encoder.layers.7.proj.weight', 'decoder.robot_encoder.layers.7.proj.bias', 'decoder.robot_encoder.layers.7.position_encoder.freqs', 'decoder.robot_encoder.layers.7.position_encoder.mlp.0.weight', 'decoder.robot_encoder.layers.7.position_encoder.mlp.0.bias', 'decoder.robot_encoder.layers.7.position_encoder.mlp.2.weight', 'decoder.robot_encoder.layers.7.position_encoder.mlp.2.bias', 'decoder.robot_encoder.layers.10.weight', 'decoder.robot_encoder.layers.11.layers.0.0.weight', 'decoder.robot_encoder.layers.11.layers.0.0.bias', 'decoder.robot_encoder.layers.11.layers.1.weight', 'decoder.robot_encoder.layers.11.layers.1.bias', 'decoder.robot_encoder.layers.12.weight', 'decoder.robot_encoder.layers.13.q_proj.weight', 'decoder.robot_encoder.layers.13.q_proj.bias', 'decoder.robot_encoder.layers.13.k_proj.weight', 'decoder.robot_encoder.layers.13.v_proj.weight', 'decoder.robot_encoder.layers.13.v_proj.bias', 'decoder.robot_encoder.layers.13.proj.weight', 'decoder.robot_encoder.layers.13.proj.bias', 'decoder.robot_encoder.layers.13.position_encoder.freqs', 'decoder.robot_encoder.layers.13.position_encoder.mlp.0.weight', 'decoder.robot_encoder.layers.13.position_encoder.mlp.0.bias', 'decoder.robot_encoder.layers.13.position_encoder.mlp.2.weight', 'decoder.robot_encoder.layers.13.position_encoder.mlp.2.bias', 'decoder.robot_encoder.layers.16.weight', 'decoder.robot_encoder.layers.17.layers.0.0.weight', 'decoder.robot_encoder.layers.17.layers.0.0.bias', 'decoder.robot_encoder.layers.17.layers.1.weight', 'decoder.robot_encoder.layers.17.layers.1.bias', 'decoder.robot_encoder.layers.18.weight', 'decoder.robot_encoder.layers.19.q_proj.weight', 'decoder.robot_encoder.layers.19.q_proj.bias', 'decoder.robot_encoder.layers.19.k_proj.weight', 'decoder.robot_encoder.layers.19.v_proj.weight', 'decoder.robot_encoder.layers.19.v_proj.bias', 'decoder.robot_encoder.layers.19.proj.weight', 'decoder.robot_encoder.layers.19.proj.bias', 'decoder.robot_encoder.layers.19.position_encoder.freqs', 'decoder.robot_encoder.layers.19.position_encoder.mlp.0.weight', 'decoder.robot_encoder.layers.19.position_encoder.mlp.0.bias', 'decoder.robot_encoder.layers.19.position_encoder.mlp.2.weight', 'decoder.robot_encoder.layers.19.position_encoder.mlp.2.bias', 'decoder.robot_encoder.layers.22.weight', 'decoder.robot_encoder.layers.23.layers.0.0.weight', 'decoder.robot_encoder.layers.23.layers.0.0.bias', 'decoder.robot_encoder.layers.23.layers.1.weight', 'decoder.robot_encoder.layers.23.layers.1.bias', 'decoder.robot_encoder.layers.24.weight', 'decoder.layers.0.adaLN_modulation.1.w [... cut ...] ', 'decoder.layers.53.4.weight', 'decoder.layers.53.4.bias', 'neck.extra_convs.0.conv.weight', 'neck.extra_convs.0.conv.bias', 'neck.extra_convs.0.gn.weight', 'neck.extra_convs.0.gn.bias', 'neck.convs.0.conv.weight', 'neck.convs.0.conv.bias', 'neck.convs.0.gn.weight', 'neck.convs.0.gn.bias', 'neck.convs.1.conv.weight', 'neck.convs.1.conv.bias', 'neck.convs.1.gn.weight', 'neck.convs.1.gn.bias', 'neck.convs.2.conv.weight', 'neck.convs.2.conv.bias', 'neck.convs.2.gn.weight', 'neck.convs.2.gn.bias']