HorizonRobotics/RoboOrchardLab #18: Training loss spike in put_bottles_dustbin task

Fanqyu, 2025-08-25:
Hi Horizon Robotics Team,

Thank you for your excellent open-source baseline for RoboTwin!

During training on the **put\_bottles\_dustbin** task, I noticed that the training loss suddenly increases sharply after reaching a certain step, then oscillates, and finally stabilizes at a relatively large value. I’d like to ask whether you have also observed this phenomenon. If so:

1. Does it affect the performance?
2. Have you figured out the reason behind this phenomenon?

I’ve attached the loss curve below. Looking forward to your response.

Best regards,
Qingyu

[image]

Fanqyu, 2025-08-25:
Thank you for your reply!
Could you please suggest a reasonable range for the weight? The original loss_depth weight is set to 1.

linxuewu, 2025-08-25:
0.1~0.3 might be a good choice.

closed 2026-04-07 by AlanHuangT