HorizonRobotics/RoboOrchardLab #19: 复现FineGrasp

liuzengyun, 2025-09-02:
地平线robo_orchard_lab团队、Yun Du同学，您好，

非常感谢您在社区的开源工作！我想follow贵团队的FineGrasp工作，但是目前复现出来的性能一直不太理想，可以麻烦帮忙看一下可能存在的原因嘛？

附：

1. 实验结果

|                   | AP    | AP Seen | AP Similar | AP Novel |
| ----------------- | ----- | ------- | ---------- | -------- |
| 论文结果           | 53.97 | 71.67   | 62.83      | 27.40    |
| 训练测试-lr=0.004  | 50.79 | 68.65   | 58.71      | 25.00    |
| 训练测试-lr=0.008  | 50.56 | 68.64   | 58.28      | 24.76    |
| 训练测试-lr=0.001  | 51.71 | 69.09   | 60.42      | 25.63    |
| 训练测试-lr=0.002  | 50.99 | 68.46   | 59.58      | 24.94    |
| 训练测试-lr=0.0005 | 50.92 | 66.72   | 60.06      | 25.98    |

1. 实验配置

硬件4张4090卡，独立实验，没有其他进程干扰。软件PyTorch=2.5.0+CUDA=11.8。

```Python
def get_trainer_cfg():
    trainer_cfg = TrainerParam(
        max_epoch=10,
        save_epoch_freq=1,
        epoch_eval_freq=100,
        step_log_freq=25,
        data_root="data/graspnet1b",
        train_split="train",
        test_split="test",
        test_mode=["test", "test_scale"],
        lr=0.004, #多次改变学习率进行实验
        train_batch_size=4,
        grad_clip_mode="norm",
        grad_max_norm=10,
        lr_scheduler_step_at="step",
    )
    return trainer_cfg
    
    
model_cfgs = FineGraspConfig(
        model_name="FineGrasp",
        seed_feat_dim=512,
        graspness_threshold=0.1,
        grasp_max_width=0.1,
        num_depth=4,
        num_view=300,
        num_angle=12,
        num_seed_points=1024,
        voxel_size=0.005,
        cylinder_radius=0.07,
        cylinder_groups=[0.25, 0.5, 0.75, 1.0],
        use_normal=False,
        loss=loss,
    )

metric_cfgs = MetricParam(
        voxel_size_cd=0.01,
        collision_thresh=0.01,
        # collision_thresh=-1,
        camera="realsense",
        test_mode=trainer_cfg.test_mode,
        num_test_proc=90,
        grasp_max_width=model_cfgs.grasp_max_width,
        num_seed_points=model_cfgs.num_seed_points,
        data_root=trainer_cfg.data_root,
    )
```

liuzengyun, 2025-09-02:
@Muyun99 非常感谢您的回复！实验的config确实有一些出入，我修改试试。另外，我的graspness标签是用EconomicGrasp的代码生成的，可能也不一样？请问您是用https://github.com/HorizonRobotics/robo_orchard_grasp_label/blob/master/examples/generate_scene_label.py这个文件生成的标签吗？非常感谢！

liuzengyun, 2025-09-02:
@Muyun99 非常感谢您的回复！实验的config确实有一些出入，我修改试试。另外，我的graspness标签是用EconomicGrasp的代码生成的，可能也不一样？请问您是用HorizonRobotics/robo_orchard_grasp_label/examples/generate_scene_label.py这个文件生成的标签吗？非常感谢！

liuzengyun, 2025-09-04:
@Muyun99 非常抱歉，再麻烦问您一下。我排查后发现确实没有使用实例级normal这一分支，只是用了第一支路的点云。于是，我重新处理了数据标签，并且打开了use_normal这一配置。然后遇到了无法正常测试的问题：
1. 10epoch训练过程是正常的，测试的时候会走到[robo_orchard_lab/models/finegrasp/finegrasp.py#L206](https://github.com/HorizonRobotics/robo_orchard_lab/blob/master/robo_orchard_lab/models/finegrasp/finegrasp.py#L206)这一行，导致后续过程报错。
2. 只训练1epoch的时候，可以正常测试。

请问您有遇到过这个问题吗？

liuzengyun, 2025-09-05:
@Muyun99 拿训练集测试了一下，也会走到这一行，graspness_score大多在0.0几，但是训练的过程和loss看起来都是正常的。use_normal关掉确实能跑，我一开始就是关着的，性能比较低。

[image]

另外，麻烦问下，我是没有这个GraspNetData/scenes/scene_xxxx/realsense/normal路径的文件的，所以仿照infer.py里的方式用以下代码生成的normal数据，不知道是不是处理的不对。（还有数值➗255的版本也试过）

[image]

Muyun99, 2025-09-08:
@liuzengyun 哈喽同学，normal 的生成这块我们尝试了两种处理方式，表现类似，一种是 infer.py 中的 Open3D 版本，另外一种比较复杂一些，也附上给你做参考，可以尝试附件的版本重新生成一版 normal 训练试试，这版 normal 生成的效果如图所示

[image]
[compute_normal_real.py](https://github.com/user-attachments/files/22200930/compute_normal_real.py)
[draw_normal_real.py](https://github.com/user-attachments/files/22200931/draw_normal_real.py)

liuzengyun, 2025-09-15:
@Muyun99 好的作者大大，最近因为我只有4卡4090，所以我调整了学习率，并且为了网络能收敛，加了一个epoch，目前整体性能能复现出来，但是会有1%以内的上下波动，我认为这是正常的。
我会尽快close这个issue。