HorizonRobotics/RoboOrchardLab #22: Question on `cam2world_gl` vs `extrinsic_cv` in SEM RoboTwin real-robot dataset

Chloe-Yan-3465, 2025-09-15:
Hello HorizonRobotics team,

I have a question regarding the real-robot RoboTwin dataset, specifically the meaning of:

- `observation/left_camera/cam2world_gl`
- `observation/left_camera/extrinsic_cv`

Previously, I thought `cam2world_gl` represents the transform between the camera and the base_link (0,0,0) point defined in the URDF. Is that correct? 

Could you clarify the correct understanding of `cam2world_gl` and `extrinsic_cv`? 

Specifically: 
- What **World Coordinate System** or **Base Frame** are they based on? 
- Is it the URDF base_link (0,0,0) or `left_link6` (where the right_camera is fixed)?

My urdf file:

[image]

[image]

Example (from `left_camera` in `place_empty_cup` task):

```text
=== cam2world_gl ===
[[[ 9.9573857e-01, -3.0173045e-03, -9.2171371e-02, -1.7736199e-01],
  [ 3.6063038e-02,  9.3261611e-01,  3.5906354e-01,  1.0416870e-01],
  [ 8.4877104e-02, -3.6085740e-01,  9.2875069e-01,  9.3343174e-01],
  [ 0.0000000e+00,  0.0000000e+00,  0.0000000e+00,  1.0000000e+00]], ... ]

=== extrinsic_cv ===
[[[ 9.9573863e-01,  3.6063075e-02,  8.4877074e-02,  9.3622595e-02],
  [ 3.0173063e-03, -9.3261617e-01,  3.6085749e-01, -2.3915121e-01],
  [ 9.2171311e-02, -3.5906351e-01, -9.2875075e-01,  9.2067635e-01]], ... ]

Thank you for your clarification!

AlanHuangT, 2025-09-16:
@Muyun99 为什么会有“ cam2world_gl” ？

Muyun99, 2025-09-16:
@Chloe-Yan-3465 oh sorry，我发现我表述错误了，中间相机标的是 middle_camera_2_left_base，标定流程产出的是一个不变量。所以对于你第一个问题：
- 手上相机标定结果是相机到 EE
- 全局相机是相机到某一个静止的坐标系（我们这边选择的是左臂的 base link）

2. LMDB dataset 解析出来是 left_camera到left_link6的外参吗？

这个涉及到标定流程和实际使用的区别。标定流程得到的是一个不变量，在实际使用的时候，我们需要统一相机外参到 world 坐标系，
- 对于全局相机，和标定结果一致，是一个定值
- 对于手上相机，需要通过 EE Pose（ee2world）以及 标定结果（cam2ee）得到（cam2world）

这个流程在打包过程中我们已经转过了，所以解析出来的三个相机的外参都是 cam2world

3. left_camera到left_link6的标定是在法兰盘的位置吗

事实上，我们是想标 left_camera 到 left_ee，对于 piper 臂来说，left_ee 的位置和 left_link6 是重合的（某些臂不一样，EE 和 TCP 的概念是有区分的）。left_link6 的位置可以从 urdf 的可视化上能够看到是在法兰盘的位置而不是夹爪中心

[image]

此外，我们会尽快放出真机数据的打包脚本供大家参考，会更加清晰一点，预计在这周内

AlanHuangT, 2025-09-16:
@Muyun99  下一个版本我们把TF信息也放到arrow里面吧， 如果来得及。我近期更新一版本库。

Chloe-Yan-3465, 2025-09-16:
明白了，非常感谢！ 
P.S. 以及感谢昨天发的标定工具